SkyNet3D/Marlin #4: 2.3.9 can't talk via usb

kgross99, 2017-05-01:
I just updated my anet board to 2.3.9 and I can't communicate with it via USB/Windows 10/Cura 15.   I can flash back to 2.3.2 fix 2 and it works fine.

oderwat, 2017-05-01:
I just (re-)tested and could not reproduce this. I thinkit is most likely a configuration issue. Try to use the exact files (with that exact configuration) from 2.3.9 with the anet-board from https://github.com/SkyNet3D/anet-board first please.

oderwat, 2017-05-03:
@ascarons22 https://github.com/SkyNet3D/Marlin/releases has all pre-releases. You need all files.

ascarons22, 2017-05-03:
Thanks for the reply, greetings and great work

oderwat, 2017-05-12:
@kgross99 Please test with 2.4.1 and report back if the problem still exists for you.

closed 2017-05-12 by oderwat