SkyNet3D/Marlin #6: How to enable manual mesh bed leveling

jrubinstein, 2017-05-13:
Following the guide here: https://github.com/MarlinFirmware/Marlin/wiki/Manual-Mesh-Bed-Leveling
to enable manual mesh bed leveling, but when I go to "level bed" using the LCD I only see "Homing XYZ". Is there something special that needs to be done to enable this feature in Skynet?

oderwat, 2017-05-14:
Which exact version of SkyNet3D are you using? If it is not 2.4.1 please consider to use that (and the anet-board from here too). The latest packaged releases (2.3.x) may all not adjust the menus for this feature. If it is the 2.4.1 I need to compile and look at this myself (my tomorrow).

jrubinstein, 2017-05-14:
Skynet : 2.4.1, anet a8
Let me know if I'm doing something wrong :)

Thanks,
James

oderwat, 2017-05-14:
You are using the (original) 5 button display I guess?

jrubinstein, 2017-05-14:
Yes, the 5 button lcd display

oderwat, 2017-05-14:
Did you enable `LCD_BED_LEVELING` too?

jrubinstein, 2017-05-14:
Yep, sure did.

oderwat, 2017-05-14:
I tried it with MESH_BED_LEVELING  and LCD_BED_LEVELING enabled and compiled for the 5 button display. This works for me.

oderwat, 2017-05-14:
Did you start "Homing XYZ"? After that it continues!

jrubinstein, 2017-05-14:
I'll try again later this morning, but it was definitely not working
yesterday. It would go to "homing xyz" and stop, I tried pressing all the
buttons and.... Nothing

jrubinstein, 2017-05-14:
Did you set probe to manual also?

oderwat, 2017-05-14:
Nope. Manual probing afaik is for auto leveling without a probe, not for the mesh leveling.

jrubinstein, 2017-05-14:
Ah, so maybe that's my problem. Will try again without that option.

jrubinstein, 2017-05-14:
Yes, that was the trick - switch off PROBE_MANUALLY and switch on
FIX_MOUNTED_PROBE. I have a new issue though - the probe points are too far
to the left of the bed (too close to X origin). How can I define the probe
points for mesh leveling?
Thanks,
James

jrubinstein, 2017-05-15:
I think I got it all sorted now.
Closing, thanks!

closed 2017-05-15 by jrubinstein