SkyNet3D/Marlin #8: Menu moves very slowly

szerwi, 2017-05-17:
Hi,
I've just updated my firmware in Anet A8 to Skynet and I have one problem. I have 5 button keypad in Anet A8 and menu moves very slowly. It takes ages to set temperature to 190 degrees for example. Is there any setting to change this speed?

oderwat, 2017-05-17:
Hold the button pressed. Does it help?

szerwi, 2017-05-17:
No, it doesn't. It moves about 2-4 degrees a second.

oderwat, 2017-05-17:
What version are you using? Current version is 2.4.3 .. see https://github.com/SkyNet3D/Marlin/releases .. 2.4.x all have a new function which accelerates buttons when being hold.

szerwi, 2017-05-17:
I have 2.3.2. Is the 2.4.3 version a beta or stable?

oderwat, 2017-05-17:
Actually 2.4.0 was the first version based on a stable Marlin release since many months :) everything before was based on Release Candidates. SkyNet3D 2.4.x are not yet available as packaged releases. I have no influence on the speed of their release besides bugging peter every other week :)

szerwi, 2017-05-17:
And one more question. In end g-code I have a command: G1 Y210 to move bed forward after the print finishes. The problem is that the bed moves very slowly to the front of the printer (it takes about 10 seconds). How to solve that? On the stock firmware it was moving in a travel speed (120mm/s) but on a Skynet firmware it moves very slowly.

szerwi, 2017-05-17:
So there's no many bugs in the latest skynet? ;)

szerwi, 2017-05-17:
And to download skynet v2.4.3 I just need to download this repository?

oderwat, 2017-05-17:
You also need the anet-board from here https://github.com/SkyNet3D/anet-board
You may look up the G1 gcode documentation http://marlinfw.org/docs/gcode/G000-G001.html
I don't know what you mean with "So there's no many bugs in the latest skynet? ;)" ... it has all bug fixed which where known and fixable for the Marlin 1.1.0-1 release. Only the most primitive could be bug free and this software is for sure very complex.

szerwi, 2017-05-17:
So should I set something like this: G1 Y210 F400?

szerwi, 2017-05-17:
In this Skynet 2.4.3 there's no configuration for Anet A8. Should I use the one from 2.3.2?

oderwat, 2017-05-17:
Sure there is .. there is in fact only a A8 configuration. You can't use the file from 2.3.2 but you can compare the values. For the real thing you should read the documentation anyway: http://marlinfw.org/docs/configuration/configuration.html

szerwi, 2017-05-17:
This config that is in Marlin folder is for Anet A8?

szerwi, 2017-05-17:
And to move bed forward faster after the print I should type G1 Y210 F400, am I right?

szerwi, 2017-05-18:
Ok, I've updated to the latest version. I have 2 more problems:
1. The bed moves very slowly to the front after the print (I use command G1 Y210)
2. The printer reseted once while was printing (don't know why. I got the printer connected to PC and using Octoprint). 
How to solve this?

oderwat, 2017-05-20:
Well those problems have nothing to do with the actual Anet port of Marlin. You may consult the Malin documentation and for the reset there are many hardware related reasons (power supply, cables, pcb, usb cable, windows "chocking") why that could happen. Look for a Anet or Marlin community for help with that.

closed 2017-05-20 by oderwat