SkyNet3D/Marlin #13: Heating Issues

Aldude, 2017-05-29:
Updated from 2.3.2 to 2.4.1 and now my hot end won't hold temperature when printing. Used Pronterface to try to see what was going on and noticed that the hot end and bed were never fully turned on but seemed to be capped at 50%. Checked Configuration.h and PID_MAX is set to 255. Not sure what else it could be.

Aldude, 2017-06-01:
I have rectified the problem with a fix I had assumed would not work going in. I set PID_MAX to 511 (2^9 - 1) which should fail a check somewhere with the processor being 8 bit and all. That being said upon checking in Pronterface the hot end is once again running at full power.

G-Pereira, 2017-06-20:
Is it resolved? If so please close the issue, so people won't get confused

oderwat, 2017-06-20:
I don't know :) I never had those problems.

closed 2017-06-20 by oderwat