SkyNet3D/Marlin #14: FILAMENT_RUNOUT_SENSOR

extremeshok, 2017-05-31:
Could the empty middle pin of S_X or S_Y connector be used for the sensor on the Anet board ?

As per the code, what would  FIL_RUNOUT_PIN be ?

````
/**
 * Filament Runout Sensor
 * A mechanical or opto endstop is used to check for the presence of filament.
 *
 * RAMPS-based boards use SERVO3_PIN.
 * For other boards you may need to define FIL_RUNOUT_PIN.
 * By default the firmware assumes HIGH = has filament, LOW = ran out
 */
//#define FILAMENT_RUNOUT_SENSOR
#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_INVERTING false // set to true to invert the logic of the sensor.
  #define ENDSTOPPULLUP_FIL_RUNOUT // Uncomment to use internal pullup for filament runout pins if the sensor is defined.
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
````

oderwat, 2017-05-31:
S_Y / S_X ? You mean the endswitch connectors? The middle pins are "+5V" for all of them. You can only use the same Pin as described for the Servo of the 3D-Touch / BLTouch... see here: https://www.thingiverse.com/thing:2091529

G-Pereira, 2017-06-20:
Is it resolved? If so please close the issue, so people won't get confused

closed 2017-06-20 by oderwat