SkyNet3D/Marlin #15: Create Anet A6, A2 config examples

extremeshok, 2017-06-04:
Current this version on skynet appears to be specifically coded for A8.
Maybe consider creating configs for A8, A6, A2 and placing them into the example directory and removing all the non relevant examples.

oderwat, 2017-06-04:
It is not coded for A8 .. it just has the "main" `Configuration.h` which fits an A8. Actually it should not remove any other examples as it is a fork and meant to be included into Marlin "soon" (tm). I for my part think that people should simply learn how to configure Marlin for their printer. I would accept some tested (and maintained) examples for those printers though, which can be included with all the others. But as I don't own a A2 or A6, I can't test them as I do with A8 and all the display variants. Sadly, there are not other reliable testers here. Not even for the A8.

extremeshok, 2017-06-04:
If your going to pull/include back into marlin, a copy of the configuration.h should be in the examples/Anet/ 
I can do a fork and pull request for A6/A2

Bikerboy96, 2017-06-07:
If anyone wants, i can send you my config for anet A8 :)

kimhee1990, 2017-06-07:
Thanks @Bikerboy96 but I have an A6

Tdoc31, 2017-06-07:
Hi Bikerboy96, wold be nice to have your A8 config file to compare with mine ?

in advance thanks

Bikerboy96, 2017-06-07:
its for front sensor mount anet a8, i changed only auto level position, you can send me yours if you have better :)

[Marlin-SkyNet3D-V2.4.4.zip](https://github.com/SkyNet3D/Marlin/files/1057557/Marlin-SkyNet3D-V2.4.4.zip)

oderwat, 2017-06-07:
There is not really a lot of things which are not individual per printer that goes into the config. The only differences between A6 and A8 are the display (trivial to change) and maybe the print bed area negative values (related to endstops). PID and other stuff should be adjusted for your own printer anyway. I don't think that this repository is a good place to swap configuration files. Especially if you just adjust sensor positions or display language or whatever. Please do this privately or use personal gist. For reference there is the original configuration documentation too: http://marlinfw.org/docs/configuration/configuration.html#configuration.h

closed 2017-06-07 by oderwat