SkyNet3D/Marlin #18: Fold into Marlin?

Bob-the-Kuhn, 2017-06-09:
It looks like the Oderwat branch is almost up to date with Marlin.  Marlin is definitely open to including  Skynet items.  PR #6020 was started earlier this year but was put off because of higher priorities.  

Now is a good time to consider folding the Skynet specific items into the main Marlin branch.

Bob-the-Kuhn, 2017-06-10:
Take a look at PR #7016 and let me know what you think.

oderwat, 2017-06-10:
What should I think? I just said that I am going to make a PR eventually. I am not ready now and I don't have time atm. I firmly tested your skynet branch but it does not work as expected. I just today updated to 1.1.3 .. I am currently not updating to the bugfix branch, because I am to busy to keep that going atm.

closed 2017-06-20 by oderwat