SkyNet3D/Marlin #19: Hotend crashes during auto bed leveling, when bed turns heat off

golobodo, 2017-06-13:
HW Settings:
- Anet A8
- RAMPS1.4, replaced Bed MOSFET to IRLB3034 (problem exists also with stock MOSFET)
- Tronxy XY-08N Sensor at 5V
- Stock PSU

SW Settings:
- tested with Skynet 2.3.2, 2.4.0 and 2.4.3
- Configuration file Skynet 2.4.3 [Configuration_Skynet_2.4.3.txt](https://github.com/SkyNet3D/Marlin/files/1071081/Configuration_Skynet_2.4.3.txt)

Description:
The problem always happens, during auto bed leveling, when the head moves down and during this time, the bed stops heating. The led of the sensor light up, but the head moves further down. Sometimes, it happens also to the X or Y endstop.

Should I use #define PIDTEMPBED or lower the #define MAX_BED_POWER? Or is there any other option to solve the problem?

benlye, 2017-06-13:
You could try disabling endstop interrupts by commenting this line in configuration.h:
`#define ENDSTOP_INTERRUPTS_FEATURE`

oderwat, 2017-06-13:
Sorry.. But this question is RAMPS1.4 related and has nothing to do with the Anet port of Marlin. Please visit the issues section of https://github.com/MarlinFirmware/Marlin for help.

closed 2017-06-13 by oderwat